DPIclimate/enviro-node #5: Allow uplink/downlink of node configuration.

dpi-mick, 2022-09-05:
Allow remote configuration of certain node configuration items. i.e. send new operating parameters to a node already in the field.

Some examples:-

**1/ Change a nodes uplink interval**

**2/ Downlink of SDI-12 configuration to node.**

- Send a set of SDI-12 commands, addresses etc to a node to have it (re)configure it's SDI-12 settings.
- May be able to design hardware to allow the node to change the addresses of connected SDI-12 sensors. i.e. isolating sensors on certain physical connectors. May even map physical connectors to SDI-12 addresses ??!!

**3/ Change MQTT broker details**

**4/ Change cellular APN.**

dajtxx, 2023-03-29:
A set of config commands can be sent to a node by publishing them to the wombat/SERIAL# topic of the MQTT broker.

The Wombat checks for these when it logs into the MQTT broker and runs them immediately.

dajtxx, 2023-05-24:
Downloading the config is sufficient to close this issue. If we want the config reported later we can open a different issue.

closed 2023-05-24 by dajtxx