DPIclimate/enviro-node #23: ESP32 clock runs fast

dajtxx, 2023-02-08:
The ESP32s clock is very inaccurate (eg 10 seconds in 40 minutes in one test) due to the default 150 kHz source. There is an option to use an internal 8.5 MHz source which requires recompiling the Arduino/ESP32 core. This is meant to be more accurate so worth a try.

The simplest fix we can put in is to connect to the internet and do a time sync using NTP or the mobile network every hour to keep this drift in check.

More complicated fixes are to either include a 32 kHz crystal on the board which takes 2 ESP32 pins, or to include a separate RTC and crystal which will need to be communicated with via I2C or SPI.

HarveyBates, 2023-02-10:
@dajtxx  
The board has an onboard 32 kHz crystal already. 
It connected to pins 8 and 9 on the schematic (https://github.com/DPIclimate/enviro-node/blob/master/electronics/motherboard/enviro-node/schematics/microcontroller.pdf).

dajtxx, 2023-02-12:
So it does! Well done :)

Now to recompile the ESP32/Arduino core. Reference:

https://github.com/espressif/esp32-arduino-lib-builder/issues/102

dajtxx, 2023-03-29:
See commit f27495d187aaae2fa22da989a00f72f47fc6b80a

The code to set the RTC source was copied from the Arduino core and modified to only use the external crystal. Given it's only feeding the RTC is should be safe to call from setup().

A clock skew adjustment parameter has also been added to the config class, and some code added to where the time is taken from the modem to try and keep the RTC in line.

closed 2023-03-29 by dajtxx