DPIclimate/enviro-node #52: Firmware logs network time year as 0123

dajtxx, 2023-07-24:
The tm.year member is 'years since 1900'. The modem gives the year as 2 digits, eg '23', and the firmware adds 100 to that to get the correct value for tm.year for use in calculating the epoch value. This is all correct and works.

There is a log message that shows the values of the tm structure and it uses the raw tm.year value, meaning it shows up as 0123:

`[ 16029][I][Utils.cpp:424] connect_to_internet(): [utils] Network time: 24/07/0123 16:06:34 40`

That log message should have 1900 added to tm.year to make it look right.

mariusCZ, 2023-07-27:
Any way to test this without the hardware? I can get my hands on an ESP32, but not on the full hardware stack.

dajtxx, 2023-07-27:
You could test it if you had a Sara R5 breakout board, like from Sparkfun. If you went that route you could hard-code the flag that causes the firmware to drop into the command line to true, ignore the various errors about missing hardware, and use the `c1 cti` command to wake the modem up and get the network time.

You'd need to ensure the ESP32 Serial1 pins are used to talk to the modem.

dajtxx, 2023-08-21:
Fixed in commit 0f0b933dda83d1d3da2a0677a87e6384d95abf0e.

closed 2023-08-21 by dajtxx